ZantFoundation/Z-Ant #62: General refactor in multiple files (Interfaces Dense Layer, Loss X, LossY... etc for working in a more ordered way)

Hitomamacs, 2024-10-10:


Maldus512, 2024-12-11:
I could take a look at this starting from the `Layer` interface but I have a question first: is it necessary to stick with the open interface approach? Is an eventual user expected to be able to define a new type of `Layer`? If not it would be simpler and more efficient to collect everything in a union, like so:

```zig
pub fn Layer(comptime T: type, allocator: *const std.mem.Allocator) type {
    return union(enum) {
        activation: ActivationLayer(T, allocator),
        dense: DenseLayer(T, allocator),
        
        pub fn forward(self: @This(), input: *tensor.Tensor(T)) !tensor.Tensor(T) {
            switch (self.*) {
                inline else => |layer| try layer.forward(input),
            }
    };
}
```

Hitomamacs, 2024-12-11:
> I could take a look at this starting from the `Layer` interface but I have a question first: is it necessary to stick with the open interface approach? Is an eventual user expected to be able to [...]

Actually not, layer would be added just by the core team. So feel free to use an Union

Hitomamacs, 2024-12-11:
Right now we decided to go  with the open interface just because it seemed more elegant. If you wanna see ho we implement other layers look at the convolutional branch

MirkoCalvi, 2024-12-11:
Hi @Maldus512, we started with that implementation first, but we had some problems. First it forced you to implement in the uniono(enum) all the methods necessary for each layer implementation. So for example when calling for ActivationLayer.get_activation() we must put it into the enum class. There's an highre risk of boilerplate code. Actually there were also another reason related to comptime variables and type, since we are using a lot of non correlated types, but rn I can't remember it. Maybe it what about the fact that in a switch the case must be comptime known.
You can see the layer refactor in commit  [08b0](https://github.com/ZIGTinyBook/Z-Ant/commit/08b0e7ea404db59b69a70d1a3b1cfe1f4c8b1fb5).
I like the idea of having a total control on types with more solid union(enum) interface, feel free to test it. 
Meanwhile an open interface gives you more flexibility with a modular and generic code. At the moment the problem with the open interface are the pointers to dynamic types.

Maldus512, 2024-12-12:
Thanks, I understand.
Two more questions: 

 1. The implementation of the `Layer` interface in specific instances could use the instance type as first parameter rather than `*anyopaque` (for example, `pub fn init(self: *Self, n_inputs: usize, n_neurons: usize) !void` instead of `pub fn init(ctx: *anyopaque, n_inputs: usize, n_neurons: usize) !void`) so that those methods can be called via dot notation directly from the instance. It would require a little of plumbing when creating the layer:
 ```zig
             return Layer(T, allocator) {
                .layer_type = layer_type,
                .layer_ptr = instance,
                .layer_impl = &.{
                    .init = struct { // create an anonymous function with the correct signature for the vtable
                        fn init(ctx: *anyopaque, n_inputs: usize, n_neurons: usize) !void {
                            const self: instanceType = @ptrCast(@alignCast(ctx));
                            try instancePtr.init(self, n_inputs, n_neurons);
                        }
                    }.init,
// ...
 ```
  Would this be a good idea?

2. Is there any reason the `init` method for the layers doesn't just return a new initialized instance? This would avoid having to declare the instance with `undefined` fields and initialize it separately.

Overall I would aim to make the layer declaration something along these lines:

```zig
    //layer 1: 784 inputs, 64 neurons
    var layer1 = denselayer(f64, &allocator).new(784, 64); //declaration and initialization 
    try model.addLayer(Layer(f64, &allocator).create(&layer1)); // the interface is created on the spot when polymorphism is needed
```

Hitomamacs, 2024-12-12:
>                            const self: instanceType = @***@***.***(ctx));

I think that we are aiming for your last idea. Where the layer declaration
is leaner. Regarding your first question yes we could use the instance type
as first parameters.
While for question 2 there is no actual reason we are just stupid haha

closed 2025-01-18 by MirkoCalvi