ZantFoundation/Z-Ant #181: Code GEneration testing Pipeline

Hitomamacs, 2025-02-23:
### Feature Description
Develop a way to test one by one auto generated math function

### Problem Statement
Explain the problem this feature will solve or the improvement it will bring.

### Suggested Solution
Describe your ideal solution or any alternatives you’ve considered.

### Additional Context
Add any other context, designs, or screenshots related to the request.

MirkoCalvi, 2025-03-09:
I was thinking about creating a service function that given the operation name and the necessary parameters returns a ModelProto with only one node representing the mathematical operation.

I think it is important to avoid boilerplate code in the codegen tests.

 "OneOpModel()" can handle all the logic and in the tests we just have to manage the file creation/deletion and try the compilation of the generated code.

MirkoCalvi, 2025-03-24:
Writing a ModelProto for each math operation was too bug friendly and a lot of work, hard to maintain ant test. The solution we adopted consists in a Python script that generated a single node .onnx file using `onnx` and `onnxruntime` libraries with fuzzed input tensors. For each Model we also compute the output of the network and save it into a .json file so that we can compare the results obtained for the execution of autogenerated Zant code .

PR #200 (2025-03-02): Codegen Testing Pipeline
## Description

Added inital codegen testing pipeline.

Every time we execute the usual codegen command, zant generates two additional files used by tests:

- `test_{model_name}.zig` -> Contains all the relevant tests generated from its template located in `tests/codeGen`
- `model_options.zig` -> Contains all the model relevant informations.

So every time you want to add new model agnostics test you will need to edit the template file.
In the near future I will also implement a way to provide own predictions test that will related to the specific model we need.

To te [... cut]
closed 2025-03-24 by MirkoCalvi