ZantFoundation/Z-Ant #220: Complete ModelProto attributes

MirkoCalvi, 2025-03-05:
### Feature Description
All the following attributes must be parsed in ModelProto:
- opset_import, repeated OperatorSetIdProto
- metadata_props, repeated StringStringEntryProto ( see #208 first)
- **not relevant at the moment**: training_info, repeated TrainingInfoProto
- functions, repeated FunctionProto

You can find all the necessary information [here](https://github.com/onnx/onnx/blob/main/onnx/onnx.proto#L438)

Referent: @MirkoCalvi

Work on branch `feature/onnx`

MirkoCalvi, 2025-03-10:
@FilippoZanelli since you have done the sub issue #221 would you like to continue with the #222 ant to complete with this?

FilippoZanelli, 2025-03-14:
yes i'll finish it

MirkoCalvi, 2025-03-22:
solved in commit fd5ae3c3cca0ae549746e9370d1819a102ea74ab by @FilippoZanelli

closed 2025-03-22 by MirkoCalvi