ZantFoundation/Z-Ant #269: Compute output shape refactor

MirkoCalvi, 2025-03-29:
### Code Section/Module
src/codegen/shape_handler.onnx

### Current Problem
Boilerplate code and difficult mantainance

### Suggested Refactoring
At the moment each compute_OP_output_shape() computes the shape of the output in shape_handler.onnx, instead it should be computed using the same method used in the standard Tensor Math.

each `compute_OP_output_shape() ` should have the following structure:
```Zig
inline fn compute_OP_output_shape(readyNode: *ReadyNode) !void {
    var shape: []const i64 = undefined;

    if (utils.getTensorShape(readyNode.outputs.items[0].name)) |tensorShape| {
        shape = tensorShape;
    } else {
        shape = tensorMath.get_OP_output_shape(....);
    }
    // associate each output to the correct shape if more than one is present
    readyNode.outputs.items[0].shape = shape;
}
```

oss: `utils.getTensorShape(name: []const u8)` is a method that search for the tensor's shape into the .onnx representation. If present, is used, else is computed.

MirkoCalvi, 2025-03-29:
asegnee: @FilippoZanelli

FilippoZanelli, 2025-03-30:
working on it

MirkoCalvi, 2025-04-12:
After PR #284 
is still Missing the correct structure on some "compute_X_output_shape()" like:

    constant
    ReLU
    softmax
    gemm
    conv
    maxPool
    reducemean
    slice
    shape
    gather
    sigmoid
    unsqueeze_
    ceil
    clip
    identity
    leaky_relu
    longsoftmax
    matmul
    split
    resize
    resize_output_shape_generic
    neg_
    pads
    mean

@FilippoZanelli @pietrobenecchi
Complete and open PR

MirkoCalvi, 2025-04-12:
fix also "slice" and "reshape"

MirkoCalvi, 2025-04-13:
completed by @pietrobenecchi @FilippoZanelli with commit aedf018

closed 2025-04-16 by MirkoCalvi