ZantFoundation/Z-Ant #311: Pow math operators

MirkoCalvi, 2025-04-25:
Implement [Pow](https://onnx.ai/onnx/operators/onnx__Pow.html) mathematical operation.

## Rules
### ONNX standard
Every math method must respect the onnx representation.
You can find all the necessary details about onnx operators [here](https://onnx.ai/onnx/operators/index.html) !  

### Standard & lean format

The "standard version" of a method is in the format:

```  
    pub fn methodName (args) !Tensor(T) {
       ...
       outputShape = compute output shape...
       outputTensor = compute output tensor...
       ...
       methodName_lean(input, args..., output)

       return outputTensor;
    }
```

It is suggested to write a `get_methodName_oputput_shape()` to compute the shape of the output so that the same method can be called during the generation of the readyGraph.

The "lean version" of a method is in the format:

```  
    pub inline fn methodName_lean (args) void {
       ...
       ... actual computation of the output
    }
```

"lean" and "standard" are **mandatory**.
The lean version is called during the codegen and the standard version is used in testings.

fishonamos, 2025-05-28:
Hi @MirkoCalvi,

I'd like to work on implementing the Pow mathematical operator for this issue.

Plan to Solve the Issue:

I’ll start by referring to the ONNX Pow operator documentation to Implement the pow Standard Method and then Compute and validate output shape.

MirkoCalvi, 2025-05-29:
Hi @fishonamos, thank you! If you want some feedback on the way feel free to open a PR for a review