ZantFoundation/Z-Ant #421: ModuleNotFoundError: No module named 'onnx'

gmacario, 2025-09-29:
### Description

Hello and apologies in advance if this is newbie question.

I have been following the "Quick Start" section of the [README.md](https://github.com/ZantFoundation/Z-Ant/blob/main/README.md) using a Dev Container in Visual Studio Code but I keep getting the following error:

```text
vscode@cb09a43dd863:/workspaces/Z-Ant$ ./zant input_setter --path ./datasets/models/mnist-8/mnist-8.onnx --shape 1,3,224,224
Executing: python3 src/onnx/input_setter.py --path ./datasets/models/mnist-8/mnist-8.onnx --shape 1,3,224,224

Traceback (most recent call last):
  File "/workspaces/Z-Ant/src/onnx/input_setter.py", line 1, in <module>
    import onnx
ModuleNotFoundError: No module named 'onnx'
vscode@cb09a43dd863:/workspaces/Z-Ant$
```

### Steps to Reproduce the bug

1. Launch Visual Studio Code 
2. VS Code Command Palette: Git: Clone
   - Repository: Clone from GitHub
   - Repository Name: `ZantFoundation/Z-Ant`
3. When asked for "Would you like to open the cloned repository?" click **Open**
4. When asked for "Do you trust the authors of the files in this folder?" click **Yes, I trust the authors**
5. Click on **Reopen in container**
6. From VS Code, Open a bash terminal

   [image]

7. Type the following instructions (adapted from README.md)

```bash
# Verify that Zig 0.14.1+ is installed
zig version

# Print zant command-line help
./zant

# Set input shape
./zant input_setter --path ./datasets/models/mnist-8/mnist-8.onnx --shape 1,3,224,224
```

### Expected Behavior

I would expect that the above commands would complete successfully.

I also noticed that inside the Dev Container neither Python nor pip are installed:

```text
vscode@cb09a43dd863:/workspaces/Z-Ant$ which python
vscode@cb09a43dd863:/workspaces/Z-Ant$ python --version
bash: python: command not found
vscode@cb09a43dd863:/workspaces/Z-Ant$ pip --version
bash: pip: command not found
vscode@cb09a43dd863:/workspaces/Z-Ant$
```

See attached screenshot

[image]

### Actual Behavior

Describe what actually happens, including error messages if any.

### Additional Context

- Host OS: Ubuntu 24.04.3 LTS 64-bit
- Visual Studio Code Version: 1.104.1
- VS Code Remote Extension installed

gmacario, 2025-09-29:
> I also noticed that inside the Dev Container neither Python nor pip are installed:

I was partially correct ;-)

```text
vscode@c3f276d2ae71:/workspaces/Z-Ant$ python --version
bash: python: command not found
vscode@c3f276d2ae71:/workspaces/Z-Ant$ python3 --version
Python 3.10.12
vscode@c3f276d2ae71:/workspaces/Z-Ant$
```

Hitomamacs, 2025-09-29:
Thanks for the detailed report! You’re right — our Docker/Dev Container is out of date and currently missing Python/pip, so commands that call Python (like input_setter) fail.

We don’t have bandwidth to work on the container right now. If you’re up for it, please feel free to open a PR to update the Dev Container/Docker setup — it would be greatly appreciated. For now, consider the Dev Container behind the current setup.

MirkoCalvi, 2025-10-21:
Fixed with commit f5a75c8a9c6325dd0f4356509a816d1768c27d03

closed 2025-10-21 by MirkoCalvi